personalrobotics/aikido #57: Add first-class support for SE(2) and SE(3) state spaces

mkoval, 2016-04-21:
We have SE(2) and SE(3) state spaces implemented in `aikido::statespace`, but they are not fully supported by the rest of Aikido (e.g. distance metrics, interpolation methods, DART conversions). We need to add this functionality to make them first-class citizens.

@aaronjoh, @lgw903, and I discussed this offline and concluded that it is *not* possible to represent SE(2) and SE(3) as `CompoundStateSpace`s of SO(2), SO(3), and a real vector space because the group operation differs. However, we may be able to gracefully handle both of these by implementing an `AffineStateSpace` or an indirect product space. We should investigate this idea further before proceeding with implementation.

jslee02, 2018-02-08:
I've added unverified list of TODOs. :sweat_smile: Please feel free to update the list and let's track what needs to be done.

egordon, 2022-12-09:
Archiving