personalrobotics/aikido #68: Setup online API documentation

jslee02, 2016-04-22:
AFAIK, there are two ways:

* [using GitHub gh-pages branch](http://rickfoosusa.blogspot.com/2011/10/howto-use-doxygen-with-github.html): [DART uses this way](http://dartsim.github.io/dart/)
  * Pros: the version of documentation can be selected arbitrarily
  * Cons: tedious initial setup, manual upload for new documentation
* [CodeDocs](https://codedocs.xyz/)
  * Pros: easy initial setup, automatic update per github commit
  * Cons: documents are generated only for the default branch (i.e., `master`)

cdellin, 2016-04-22:
FWIW for my LEMUR docs (C++ using Doxygen), I found a way to get them built by [Read the Docs](https://readthedocs.org/), even though they technically only support [Sphinx](http://sphinx-doc.org/).  RtD has Doxygen installed on their build servers (to support projects using [Breathe](https://breathe.readthedocs.org/).  It works because Sphinx is configured by a `conf.py` script, which is just `import`-ed by Sphinx during the build process, so you can add whatever code you want to it to do arbitrary things during the build process.  In my case, I just generate an empty `contents.rst` (so Sphinx generates nothing), then shell out to call Doxygen on the project, making sure to put the output html where RtD expects the generated docs to be.

The [LEMUR docs](http://lemur-planning.readthedocs.org/) are a bit complicated, because I split it into multiple separate Doxygen projects (for ROS compatibility), so my [`conf.py` file](https://github.com/personalrobotics/lemur/blob/master/readthedocs/conf.py) file is a bit complicated (it post-processes the generated html documentation to add html headers and footers linking together the different packages).  If you just want to run Doxygen on a single project, it would be much sorter (mostly just a `subprocess.call('Doxygen')`).  I think Read the Docs supports building docs for different branches, and of course does auto-building on each commit.

psigen, 2016-04-22:
We should pick an option that still lets us build HTML documentation directly from the repository if we want to, both for flexibility of hosting and to make it sane to author the docs.

* `gh-pages` more or less hosts any static HTML, so that should work with any doc generator.
* `CodeDocs` and `ReadTheDocs` both have slightly specialized build systems that kind of lock us into particular doc systems (Doxygen and Sphinx + Breathe, respectively).  This is fine as long as we know what we are getting into, and _we can still build the same docs locally_.
* `ReadTheDocs` is nice in that its underlying hosting mechanism is open-source, as well as its theme.  It does support building per-branch out of the box.
* `ReadTheDocs` occasionally has really slow servers.  `CodeDocs` might as well.  Since we are operating on the free tier, we are getting bargain basement hosting here.  I would expect that `Github` or any static hosting provider will handily outperform these services in uptime and load-times, which significantly reduces frustration when knee-deep in API docs.
* @jslee02: how does search work on the static `gh-pages`?  Does it exist? Is it handled client-side?
* It is highly likely we will need customizations here or there to handle the complex combination of C++, Python, and autogenerated C++/Python bindings that we are building.

Before we commit to any online API, I think we should try to build the docs locally with Sphinx + Breathe and Doxygen and see how we like their respective layouts.  Let's not jump ahead and then get tied into some online framework and be unable to build/mirror our own docs.

cdellin, 2016-04-22:
Certainly +1 for what @psigen said.  For LEMUR I have a [Makefile](https://github.com/personalrobotics/lemur/blob/master/readthedocs/Makefile) so I can just `cd` into the docs directory and type `make` to generate the docs locally, and it's essential.

jslee02, 2017-10-24:
@psigen's suggestion: https://github.com/personalrobotics/aikido/pull/245#issuecomment-338445870
>  I would _highly_ recommend using a doc generator that can be run locally to produce static HTML, then use manual or conventional CI processes to put the static HTML artifact on a web server of your choosing.  GitHub Pages works really well for this, and is also an intuitive place for people to look for documentation.
> 
> This gives us the most flexibility (if we decide to change generators), reliability (we can manually build, upload, **backup**, and correct issues if needed), and customizability (we can control which documentation gets built).

brianhou, 2018-02-10:
I'm in favor of just putting what we have online now. #319 cleans up the Doxygen results so that it doesn't include paths from all over your local filesystem, so I'm now somewhat satisfied by its output.

> How does search work on the static gh-pages? Does it exist? Is it handled client-side?

I think if we use GitHub Pages (and probably these other options, if they just produce static HTML/CSS/JS) we can only do client-side search. Doxygen supports client-side search, although it's quite slow (from personal experience with the DART docs). I don't think it's worth the effort to figure out how to optimize this now, but it's definitely going to be a bit annoying.

The next step is to hook things up in CI. In the meantime, I think we can manually push versions to `gh-pages` starting with 0.3.0.

brianhou, 2018-03-06:
Not completely -- we still need to figure out how to deploy different versions. We can create a new issue to track that though.

jslee02, 2018-03-06:
👍 for creating a new issue for that.