personalrobotics/aikido #85: Move DART-specific constraints into a subdirectory

mkoval, 2016-04-23:
@cdellin pointed out that we put the DART-specific state spaces in the `aikido/statespace/dart` subdirectory, but make no such distinction for `aikido/constraint`. We should consider the same directory structure for `aikido/constraint` for consistency.

psigen, 2016-04-24:
Does it make sense to invert this grouping?  I.e. move these into:
`aikido/dart/statespace`
`aikido/dart/constraint`

This would make the DART-specific code all be in one namespace which could be a component (which many other components would probably depend on).  I am not sure if this is useful or not, but it may make some of the use-cases @cdellin was proposing easier by decoupling DART as a hard dependency.

Specifically, now that `statespace` is its own thing, it's more likely that planners and trajectory representations are be decoupled from interfacing directly with DART.

brianhou, 2018-01-25:
I'd like to do this since we've been working on cleaning up lately. I'm not sure whether it makes sense to invert this grouping, though. How useful do we think AIKIDO is without DART?

jslee02, 2018-02-08:
I would prefer not to have a subdirectory for DART-specific objects because it's AIKIDO's mandatory dependencies. The general rule of AIKIDO I believe is to have a subdirectory for an optional component that depends on optional dependencies.

PR #342 (2018-02-24): Clean up constraint namespace
- Creates the `constraint::uniform` namespace to mirror the existing directory structure
- Resolves #85 by introducing a `constraint::dart` namespace

***

**Before creating a pull request**

- [x] Document new methods and classes
- [x] Format code with `make format`

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`
- [x] Add unit test(s) for this change
closed 2018-02-24 by brianhou