personalrobotics/aikido #95: Inconsistent naming: Sampleable/SO2Sampleable/SO2UniformSampler

psigen, 2016-04-29:
Currently, the sampling interface is called `Sampleable`.
We define a header called `SO2UniformSampler`.
Inside it is a class called `SO2Sampleable`.

We need to standardize on these things.  I posit the following naming convention:
- **virtual** and **abstract** classes adopt the _adjective_ name: `Sampleable`
- **concrete** classes adopt the _noun_ name: `Sampler`

This makes some sense when doing inheritance, as the interface describe what the class will do, but if a class does multiple things it may not be named after any of them, e.g.
```
SO2Sampler : Sampleable
TSR : Sampleable, Projectable
```

psigen, 2016-04-29:
This is related to #91 but more drastic 😓

jslee02, 2017-10-01:
Resolved by #224

closed 2017-10-01 by jslee02