personalrobotics/aikido #98: Compiling without Bullet seems to cause issues.

psigen, 2016-05-05:
I have a version of DART built without Bullet, and I am getting the following error when compiling:

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
BULLET_INCLUDE_DIR
```

This might be a configuration issue in the exported DART config, I'm not sure yet.

jslee02, 2016-05-05:
This was [resolved](https://github.com/dartsim/dart/pull/697) for master. Maybe I should backport this to `dart-prl0`.

psigen, 2016-05-05:
I see, I don't need a fix right now, but it would be useful to address this whenever we next update the DART fork.

psigen, 2016-05-05:
I'm not sure if i should keep this open or not since it is an upstream issue.

jslee02, 2016-05-05:
I simply [cherry-picked the commit](https://github.com/personalrobotics/dart/commit/d17c7f4d44494365cb39a91a3ad9b73b5402895a) to `personalrobotics/dart` so you would be fine if you build and install DART from source. 

The reason I don't reupload `dart-prl0-dev` with the fix is that DART 6.0 will be released any soon once [one last PR of DART](https://github.com/dartsim/dart/pull/694) is finished, which I'm currently working on. 

We can leave this open until we get to using DART 6.0 for aikido.

psigen, 2016-05-05:
That all sounds good, thanks @jslee02!