personalrobotics/aikido #117: Consider switching to Codecov for online coverage reporting

jslee02, 2016-11-22:
The graphical user interface of [Codecov](https://codecov.io/gh/dartsim/dart/branch/codecov) looks more organized and intuitive than [Coveralls](https://coveralls.io/github/dartsim/dart?branch=master).

Related issue: [DART #806](https://github.com/dartsim/dart/pull/806)

mkoval, 2016-11-23:
👍 Codecov has a much nicer UI. I've always thought that Coveralls looks a bit like the `Coveralls` project page looks a bit like a ransom note. 😁

@psigen Thoughts?

psigen, 2016-11-23:
Generally fine with this.  But let's try it out before we commit to it completely.

We should be able to add `codecov` support in a branch without removing the `coveralls` support, right?  I think they have non-overlapping configuration.  We can try that and see how it looks?

jslee02, 2017-05-05:
I've created a PR for this. Please take a look at it when you have a chance.

closed 2017-05-11 by jslee02