personalrobotics/aikido #123: Consider using PPA for Aikido debian packages

jslee02, 2017-01-09:
Once we start versioning of Aikido, it would be also useful to upload debian packages of Aikido to a PPA. We could start a new PPA for the personal robotics packages or use [DART PPA](https://launchpad.net/~dartsim/+archive/ubuntu/ppa/+packages) for a smooth start.

jslee02, 2017-10-01:
[PRL PPA](https://launchpad.net/~personalrobotics/+archive/ubuntu/ppa) was created, and `libaikido0-*-dev` was uploaded for the current master branch. I couldn't update aikido 0.0.1, which is tagged as v0.0.1, because it couldn't be built without errors.

Once aikido 0.1.0 is released, we could update the package and close this issue.

brianhou, 2017-10-04:
Is this resolved? I noticed this repo was also created: https://github.com/personalrobotics/aikido-debian/

jslee02, 2017-10-04:
@brianhou Yes, this is resolved. To be clear, [`aikido-debian`](https://github.com/personalrobotics/aikido-debian/) is to manage debian files, that are required to update packages to a PPA, and the actual PPA is at here: https://launchpad.net/~personalrobotics/+archive/ubuntu/ppa.

Closing this issue.

closed 2017-10-04 by jslee02