personalrobotics/aikido #128: aikido rviz always fails

vinitha910, 2017-01-12:
aikido rviz always fails silently because of dependencies (libmicrohttpd)

psigen, 2017-01-13:
Perhaps now would be a good time to think a bit if a header-only implementation would be useful here, since the requirements are relatively simple (serving static content).

https://github.com/yhirose/cpp-httplib

jslee02, 2017-04-27:
I believe we could close this issue because the dependency of `libmicrohttpd` is explicitly added in #149, and `rviz` component is being built in CI testing.

@psigen Do you mean making Aikido header-only project? If so, I think we should create an issue for that separately.

psigen, 2017-04-27:
No, I mean that we could use a header-only implementation of an HTTP server so that we do not have to depend on libmicrohttpd, which is not exactly common and adds another non-standard rosdep key.

jslee02, 2017-04-27:
Oh, I see. I don't have an objection to using a header-only library as long as it meets our requirements and is mature enough.

jslee02, 2017-04-27:
I created an issue for using a header-only library to continue the discussion there. Closing this issue.

closed 2017-04-27 by jslee02