personalrobotics/aikido #132: OMPL Dependency

ClintLiddick, 2017-01-26:
While building aikido without catkin there was an error when using the Ubuntu 14.04 packaged OMPL (0.13), but the latest version (1.2.1) installed from source did not work either (many breaking changes, and also requiring a fully c++11 complier and stdlib, so GCC5+ which is not available on 14.04). Installing OMPL 1.1.0 from source worked. I'm not sure the best way to handle this odd dependency situation, but I know we should be able to build outside of a catkin workspace.

mkoval, 2017-01-26:
DART and Aikido both use of C++11 _extensively_ (@mxgrey is a C++ wizard), so I am surprised that you are having issues building with OMPL but _not_ having issues building DART with the same compiler. Are you sure the issues aren't more mundane types of backwards incompatibility?

PyryM, 2017-03-16:
Was it this issue:
```
aikido/include/aikido/planner/ompl/GeometricStateSpace.hpp:68:10: error: ‘double aikido::planner::ompl::GeometricStateSpace::getMeasure() const’ marked ‘override’, but does not override
   double getMeasure() const override;
          ^
```

Because I am having this issue as well. This is on ubuntu 14.04 with gcc 5.4.1 (I installed gcc 5), OMPL from the package manager.

mkoval, 2017-03-17:
It looks like `getMeasure` was added in OMPL 1.0.0. The latest version or `ros-indigo-ompl` is nominally 1.0.3094, which is (I believe) build number 3094 of 1.0.0. Are you using the ROS Debian package for OMPL (i.e. `ros-indigo-ompl`) or some other build? Can you post the output of `dpkg -l | grep ompl`?

I don't think this is the same issue that @ClintLiddick had before, but I am not 100% sure.

PyryM, 2017-03-17:
This is without ROS, so it's just the ubuntu package "ompl", which is [apparently 0.13.0](http://packages.ubuntu.com/source/trusty/ompl). I'll install ompl > 1.0 from source and see if it compiles then; a note in the readme might be worthwhile.

jslee02, 2017-03-17:
I also found that OMPL switched from boost smart pointers to STL smart pointers since OMPL 1.2.0. This prevents installing Aikido on macOS when the recent version of OMPL is installed (1.3.0).

I think we should either:
(1) support OMPL 0.13.0 (for Ubuntu 14.04 LTS) and above with some bandage code and build scripts for the backward compatibility or
(2) drop supporting old version of OMPL and enforce people to install newer version when they are on Ubuntu 14.04 LTS.

cdellin, 2017-03-17:
(Hah reminds me of [OR_OMPL_HAS_BOOSTSMARTPTRS](https://github.com/personalrobotics/or_ompl/search?utf8=%E2%9C%93&q=OR_OMPL_HAS_BOOSTSMARTPTRS&type=Code))

aditya-vk, 2017-04-20:
The tests under `tests/planner/ompl` fail to build with the latest OMPL version 1.3.0. Support for OMPL 1.2.0 and above set up by @jslee02 fixes most of the issues. (Travis builds with OMPL 1.1.0 and tests build successfully). Opened a branch[WIP] `fix/tests_ompl_ver` for the same.

jslee02, 2018-02-08:
I don't believe this issue is still persistent. Please reopen if it's not.

closed 2018-02-08 by jslee02