personalrobotics/aikido #145: Disable Reviewable integration

mkoval, 2017-03-28:
[Github now has a built-in code review system](https://github.com/features) that integrates nicely with the UI and branch protection. While this system isn't perfect, it does seem to be quite a bit more polished than Reviewable. I would like to disable Reviewable on Aikido and switch to using Github code review.

@psigen @jslee02 Any objections?

psigen, 2017-03-28:
I am in support of this, I have not been using Reviewable lately because it
is very slow to load for me.

jslee02, 2017-03-28:
Yeah, we've used it because there was no alternative that allows us to review the whole diff when it's large. The GitHub's built-in review system looks good enough for our use to me.

brianhou, 2017-04-02:
Closing -- I flipped the switch on the Reviewable page.

closed 2017-04-02 by brianhou