personalrobotics/aikido #181: Consider splitting utils into common and io

jslee02, 2017-04-19:
The current `utils` component seems to play two roles: 
(1) common utilities used by most of other Aikido components
(2) file input-output utilities

I believe we want to keep the common component as light as possible regarding required external dependencies. For this, I would like to suggest splitting it into `common` and `io` components as
(1) `common`: a default (build always) component that requires only the essential dependencies (such as Eigen, DART).
(2) `io`: an optional component that probably requires additional dependencies related to file input-output such as `tinyxml2` and `yaml-cpp`.

mkoval, 2017-04-21:
I like this idea. 👍

jslee02, 2017-10-01:
Resolved by #225

closed 2017-10-01 by jslee02