personalrobotics/aikido #220: StepSequnce test fails on 32-bit Ubuntu

jslee02, 2017-09-27:
Here is the test error message:
```
/<<PKGBUILDDIR>>/tests/util/test_StepSequence.cpp:94: Failure
Value of: seq2.getMaxSteps()
  Actual: 5
Expected: 6
[  FAILED  ] StepSequence.MaxSteps (0 ms)
[ RUN      ] StepSequence.MaxStepsAlternateRanges
/<<PKGBUILDDIR>>/tests/util/test_StepSequence.cpp:109: Failure
Value of: seq2.getMaxSteps()
  Actual: 20
Expected: 21
```

I found this test failure working on uploading Aikido to the PRL PPA ([build log](https://launchpadlibrarian.net/338903511/buildlog_ubuntu-trusty-i386.aikido0_0.0.2.ppa1~trusty_BUILDING.txt.gz)).

yskim041, 2017-10-03:
aikido 0.1.0 got same error messages and one more from `test_RosTrajectoryConversions`.

```
[ RUN      ] StepSequence.MaxSteps
/<<PKGBUILDDIR>>/tests/common/test_StepSequence.cpp:94: Failure
Value of: seq2.getMaxSteps()
  Actual: 5
Expected: 6
[  FAILED  ] StepSequence.MaxSteps (0 ms)
[ RUN      ] StepSequence.MaxStepsAlternateRanges
/<<PKGBUILDDIR>>/tests/common/test_StepSequence.cpp:109: Failure
Value of: seq2.getMaxSteps()
  Actual: 20
Expected: 21
[  FAILED  ] StepSequence.MaxStepsAlternateRanges (0 ms)
...
[ RUN      ] ToRosJointTrajectoryTests.TrajectoryHasCorrectWaypoints
/<<PKGBUILDDIR>>/tests/control/ros/test_RosTrajectoryConversions.cpp:145: Failure
Value of: rosTrajectory2.points[i].time_from_start.toSec()
  Actual: 0.029999999000000003
Expected: timestep*i
Which is: 0.029999999999999999
[  FAILED  ] ToRosJointTrajectoryTests.TrajectoryHasCorrectWaypoints (0 ms)
```

brianhou, 2018-01-25:
I think @dqyi11 fixed these tests in in #305, but let's keep an eye out next time that we package AIKIDO.

jslee02, 2018-01-25:
Can we reopen this issue and close once we're sure if it's fixed in the next packaging?

jslee02, 2018-02-11:
This is still persistent with the current master (https://github.com/personalrobotics/aikido/tree/6f921558e50a04ae3d8730282d7b5bdf827ab411) (see this [build log](https://launchpadlibrarian.net/356589123/buildlog_ubuntu-trusty-i386.aikido0-prerelease_0.2.0.201802110406+a91dcfd~trusty~ubuntu14.04.1_BUILDING.txt.gz)).

brianhou, 2018-02-11:
Are we sure that the new implementation is correct? Why is the old precision logic with `std::abs` no longer necessary? (It seems that https://github.com/personalrobotics/aikido/blob/master/src/common/StepSequence.cpp#L9 isn't used at all.)

dqyi11, 2018-02-11:
I will take a look at this issue. All the tests passed in 64-bit machine. 
The magic number is not used by intent, which shall be actually removed.
I will remove it in fixing this issue.

dqyi11, 2018-04-05:
This is caused by the difference of numeric precision between 32bit and 64bit. It will be fixed in a following PR.

dqyi11, 2018-04-20:
Resolved in PR #365

closed 2018-04-20 by dqyi11