personalrobotics/aikido #267: VectorFieldPlanner

dqyi11, 2017-11-26:
VectorField planner that plans a path by following defined joint velocities in configuration space and terminates by defined criteria.
- Use `boost::numeric::odeint::integrate_adaptive` that integrates joint velocities with adaptive step size
- Refactor implementation to follow the design in `prpy`;
[https://github.com/personalrobotics/prpy/blob/master/src/prpy/planning/vectorfield.py](https://github.com/personalrobotics/prpy/blob/master/src/prpy/planning/vectorfield.py)
- Add three vector fields 
  - [MoveEndEffectorOffset] that moves an end-effector to an offset by a direction;
  - [MoveEndEffectorPose] that moves an end-effector to a goal pose;
  - [MoveEndEffectorPoseAlong] that moves an end-effector along a given workspace trajectory.

jslee02, 2018-01-06:
@dqyi11 All the issues here resolved by #268? If so, could you please close this?

dqyi11, 2018-01-06:
Not yet, there will be another PR for 
> [MoveEndEffectorPoseAlong] that moves an end-effector along a given workspace trajectory.

jslee02, 2018-01-06:
I added a checklist to the description accordingly. Feel free to modify if necessary.