personalrobotics/aikido #269: Travis CI is silent even when it fails to build AIKIDO

jslee02, 2017-11-27:
See [this build](https://travis-ci.org/personalrobotics/aikido/jobs/308102385#L1846-L1856)

brianhou, 2017-11-27:
I think the silent failing issue may be because that we're missing `set -ev` in all of our `.ci` scripts?

jslee02, 2017-11-27:
What `set -ev` does do?

brianhou, 2017-11-27:
`-e` causes the script to exit immediately after a command fails, which I think will set the error code correctly for Travis. `-v` is just to be verbose and include the commands that are being run, which might be helpful for debugging.

I think we should merge https://github.com/personalrobotics/pr-cleanroom/pull/7 first. If that doesn't cause Travis to fail, we should then see if we need to add `-e` to the `.ci` scripts in #270.

jslee02, 2017-11-27:
Thanks for the explanation!

Let's close this issue once it's confirmed if personalrobotics/pr-cleanroom#7 actually fixes this issue. Reopening for now.

brianhou, 2017-12-08:
Resolved by #279.

PR #270 (2017-11-28): Add set -ev to all .ci scripts.
Maybe resolves #269?

***

**Before creating a pull request**

- [x] Document new methods and classes (NA)
- [x] Format code with `make format` (NA)

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side (NA)
- [x] Summarize this change in `CHANGELOG.md` (NA)
- [x] Add unit test(s) for this change (NA)
closed 2017-12-08 by brianhou