personalrobotics/aikido #284: Consider adding name field to Trajectory

jslee02, 2017-12-09:
I think it would be good to be able to assign a name to a trajectory. This would be useful to know which trajectory is which when visualizing multiple trajectories in a viewer (e.g., rviz). This can be done by adding a name field to the associated visualization object (e.g., `TrajectoryMarker`), but I think it would be worth to have them in `Trajectory` class.

Thought?

psigen, 2017-12-10:
In general, I am a little wary of assigning metadata like names to pure data structures like `Trajectory`.  It makes it a bit awkward in situations like:

1) A trajectory is being created and consumed inside an optimizer.  What should the name be?
2) A trajectory is being stored into a database by ID.  Does the name have to be synchronized with the database ID when it is saved/loaded/copied?  Can the name be a binary string (since that might be its unique identifier)?
3) A trajectory is cloned a number of times.  Should its name change?  Does the name need to be unique?

With that said, it can be user-friendly to keep the name around as a metadata field for debugging. If it is really necessary to maintain a name, absolutely _nothing_ in DART or its plugins should assume the field:
1) is populated
2) is unique
3) is meaningful

mkoval, 2017-12-12:
I agree with @psigen here. I generally like the idea of keeping name in visualization classes for all of the reasons he outlined above.

However, I do think it would be tremendously useful to other types of metadata with a trajectory, e.g. how it was planned, what constraints it was checked against, and where those constraint checks were performed. This is very similar to the concept of "trajectory tags" we had in PrPy that @brianhou began to resurrect in https://github.com/personalrobotics/aikido/pull/259.

---

This is _precisely_ the problem that DART `Aspect`s https://github.com/dartsim/dart/pull/659 were intended to solve. I would be thrilled if we added `Aspect`s to Aikido's `Trajectory` class. I don't see any issue with creating a `VisualizationAspect` that could hold the trajectory's display name.