personalrobotics/aikido #311: StepSequence test case

dqyi11, 2018-01-23:
I have a few questions about the test cases for `StepSequence` class, which is related with the design of the class.

(1) In the docstring of the constructor, `_includeEndpoints` means
>If includeEndpoints is true then the final point in the sequence will be the end point, even if it is at less than stepSize from the second to last point.

But in one test case, it seems that `_includeEndpoints` also applies to start point. Because we have
> StepSequence seq(0.55, true);
>  EXPECT_DOUBLE_EQ(0.0, seq[-1]);

Shall we change to `_includeEndpoint` (no s) and throw a `out_of_range` exception in this case?
An alternative is having both `includeStartpoint` and `includeEndpoint` like in `VanDerCorput` class.

(2) The test for iterator is not fully covered.
There was only `StepSequence seq(0.3, true)` and `StepSequence seq(0.2, false)` is added recently.
The current code will still fail in `StepSequence seq(0.3, false)`.

I have created a branch in fixing this but want to confirm the requirement firstly.

brianhou, 2018-01-25:
It seems from #52 that the intent for `StepSequence` was to be similar to `VanDerCorput`, so let's mirror the API change that was introduced in #37. @dqyi11, can you make this change and add tests in your branch?

Since we first stumbled across these issues as part of #303, please make sure that those tests pass as well before making a PR.

jslee02, 2018-02-08:
I believe this is resolved by #320.

closed 2018-02-08 by jslee02