personalrobotics/aikido #312: Create a WaypointTrajector class

dqyi11, 2018-01-24:
Currently both the `Spline` class and the `Interpolated` class have a few methods for "waypoints", which are `getNumWaypoints()`, `getWaypoint(i)` and `getWaypointTime(i)`, which are not inherited from the base class `Trajectory`.
Those methods are mostly called in converting to some implicit path format, for example `DynamicPath` in `computeParabolicTiming()`. 

@brianhou  suggested creating an abstract class `WaypointTrajectory` (in between base class `Trajectory` and `Interpolated` \ `Spline`) and defining all the "waypoints" operations in this class.
The change will merge conversions based on "waypoints" from both `Interpolated` and `Spline` into a single method, which will remove duplicate code.

A very minor concern is that whether it is fair to consider `Spline` as a `WaypointTrajectory`.
Because `Spline` is a sequence of line segments. 
The "waypoints" of a `Spline` is "indirectly" obtained from segments.
https://github.com/personalrobotics/aikido/blob/master/src/trajectory/Spline.cpp#L206
https://github.com/personalrobotics/aikido/blob/master/src/trajectory/Spline.cpp#L226

We would like to hear some comments before making the change.

psigen, 2018-01-25:
Quick question: Can `Spline` be implemented as a type of `interpolator` for `Interpolated`?

brianhou, 2018-01-25:
I was wondering the same thing! After some investigation, I think the (current) answer is no because each segment of the spline has different polynomial coefficients and `Interpolator::interpolate(State* from, State* to, double alpha, State* state)` doesn't provide a way to pass those in.

Maybe it would be possible to model a `Spline` trajectory as a sequence of `Interpolated` segments, where each segment has its own `PolynomialInterpolator`? I'm not sure if that would be useful either.

psigen, 2018-01-25:
Yeah, I suppose it gets a bit weird once you have each segment need different interpolation parameters.  One thing that's kind of interesting though is that if you have fixed order of the segments of the spline, and you have specified the start and end waypoints to the `order/2`-th derivative, then this is a fully constrained solution and you no longer need the polynomial coefficients.

Our existing Spline class is somewhat more general here, but it's interesting that for a certain subset of waypoints and fixed-order splines, this should be do-able.

mkoval, 2018-02-02:
> Maybe it would be possible to model a `Spline` trajectory as a sequence of Interpolated segments, where each segment has its own `PolynomialInterpolator`? I'm not sure if that would be useful either.

This is the way I would approach it, if we decide to go this way. The spline parameters are an implementation detail of the interpolator - not a formal argument required to evaluate the trajectory.

The bigger issue is that `Interpolated` and `Interpolator` currently only allows zero velocity waypoints. We often create `Spline`s with non-zero velocity at waypoints, e.g. inside the Hauser parabolic smoother. If we wanted to do this, we would have to relax this limitation. E.g. change the signature to:

I don't see a huge amount of value in this change, so I suggest leaving things as they are for now.

closed 2018-02-02 by brianhou