personalrobotics/aikido #348: Remove dtwarn redefinition

brianhou, 2018-02-23:
https://github.com/personalrobotics/aikido/pull/342#discussion_r170391883

There are a few times that `dtwarn` is redefined in `constraint::dart` because the namespace `dart` is ambiguous. We should remove them once they're no longer necessary.

jslee02, 2018-02-24:
Upstream fix: https://github.com/dartsim/dart/pull/1010

brianhou, 2019-09-03:
@jslee02 I think we should be able to update the DART version we depend on now, so we'll have that upstream fix. Can you try to address this? Thanks!

jslee02, 2019-09-04:
I believe the redefinition warnings should be gone by now because AIKIDO is already using the DART version with the fix. The fix is in DART 6.4, and [AIKIDO depends on DART >= 6.6.2](https://github.com/personalrobotics/aikido/blob/ec943527925fbfdc4082aedf5f155883dcdcf095/CMakeLists.txt#L90).

PR #550 (2019-09-10): Remove dtwarn redefinitions.
Resolves #348.
closed 2019-09-10 by brianhou