personalrobotics/aikido #363: Segfaults with OMPL 1.2.0 or higher

jslee02, 2018-03-22:
[All of AIKIDO's tests for OMPL segfault with OMPL 1.2.0](https://ci.appveyor.com/project/jslee02/aikido/build/91/job/37puse8agadeecb5#L5845) or higher. Here is the build configuration:
* OS: Ubuntu Xenial
* Compiler: gcc 5.4.0
* Boost: 1.59
* OMPL: 
  * working: 1.0.0, 1.1.1
  * now-working: 1.2.0, 1.2.1, 1.3.0, 1.3.1, 1.3.2

I have tested different versions and binary packages (`libompl-dev` and `ros-indigo/kinetic/lunar-ompl`) and ones built from source, but only OMPL 1.2.0 or higher have the issue.

I couldn't pinpoint a line causes the segfault using a debugger. The debugger just points assembly code as
```asm
0x7ffff7de7620                   41 56                    push   %r14
0x7ffff7de7622  <+0x0002>        41 55                    push   %r13
0x7ffff7de7624  <+0x0004>        41 54                    push   %r12
0x7ffff7de7626  <+0x0006>        55                       push   %rbp
0x7ffff7de7627  <+0x0007>        53                       push   %rbx
        36 [1]	in dl-init.c
0x7ffff7de7628  <+0x0008>        0f b6 87 14 03 00 00     movzbl 0x314(%rdi),%eax
        39 [1]	in dl-init.c
0x7ffff7de762f  <+0x000f>        4c 8b 47 08              mov    0x8(%rdi),%r8
        36 [2]	in dl-init.c
0x7ffff7de7633  <+0x0013>        83 c8 08                 or     $0x8,%eax
0x7ffff7de7636  <+0x0016>        88 87 14 03 00 00        mov    %al,0x314(%rdi)
        39 [2]	in dl-init.c
0x7ffff7de763c  <+0x001c>        45 0f b6 08              movzbl (%r8),%r9d
0x7ffff7de7640  <+0x0020>        45 84 c9                 test   %r9b,%r9b
0x7ffff7de7643  <+0x0023>        0f 84 cf 00 00 00        je     0x7ffff7de7718 <call_init+248>
        44 [1]	in dl-init.c
0x7ffff7de7649  <+0x0029>        48 8b 87 a0 00 00 00     mov    0xa0(%rdi),%rax
0x7ffff7de7650  <+0x0030>        49 89 cd                 mov    %rcx,%r13
0x7ffff7de7653  <+0x0033>        49 89 d4                 mov    %rdx,%r12
0x7ffff7de7656  <+0x0036>        89 f5                    mov    %esi,%ebp
0x7ffff7de7658  <+0x0038>        48 89 fb                 mov    %rdi,%rbx
0x7ffff7de765b  <+0x003b>        48 85 c0                 test   %rax,%rax
0x7ffff7de765e  <+0x003e>        74 70                    je     0x7ffff7de76d0 <call_init+176>
        49 [1]	in dl-init.c
0x7ffff7de7660  <+0x0040>        f6 05 39 56 21 00 02     testb  $0x2,0x215639(%rip)        # 0x7ffff7ffcca0 <_rtld_global_ro>
0x7ffff7de7667  <+0x0047>        75 7f                    jne    0x7ffff7de76e8 <call_init+200>
        58 [1]	in dl-init.c
0x7ffff7de7669  <+0x0049>        48 8b 40 08              mov    0x8(%rax),%rax
0x7ffff7de766d  <+0x004d>        48 03 03                 add    (%rbx),%rax
0x7ffff7de7670  <+0x0050>        4c 89 ea                 mov    %r13,%rdx
0x7ffff7de7673  <+0x0053>        4c 89 e6                 mov    %r12,%rsi
0x7ffff7de7676  <+0x0056>        89 ef                    mov    %ebp,%edi
0x7ffff7de7678  <+0x0058>        ff d0                    callq  *%rax
0x7ffff7de767a  <+0x005a>        48 8b 93 08 01 00 00     mov    0x108(%rbx),%rdx
        62 [1]	in dl-init.c
0x7ffff7de7681  <+0x0061>        48 85 d2                 test   %rdx,%rdx
0x7ffff7de7684  <+0x0064>        74 3d                    je     0x7ffff7de76c3 <call_init+163>
        68 [1]	in dl-init.c
0x7ffff7de7686  <+0x0066>        48 8b 83 18 01 00 00     mov    0x118(%rbx),%rax
        70 [1]	in dl-init.c
0x7ffff7de768d  <+0x006d>        48 8b 1b                 mov    (%rbx),%rbx
0x7ffff7de7690  <+0x0070>        48 03 5a 08              add    0x8(%rdx),%rbx
        68 [2]	in dl-init.c
0x7ffff7de7694  <+0x0074>        48 8b 40 08              mov    0x8(%rax),%rax
0x7ffff7de7698  <+0x0078>        48 c1 e8 03              shr    $0x3,%rax
        71 [1]	in dl-init.c
0x7ffff7de769c  <+0x007c>        85 c0                    test   %eax,%eax
0x7ffff7de769e  <+0x007e>        74 23                    je     0x7ffff7de76c3 <call_init+163>
0x7ffff7de76a0  <+0x0080>        83 e8 01                 sub    $0x1,%eax
0x7ffff7de76a3  <+0x0083>        4c 8d 74 c3 08           lea    0x8(%rbx,%rax,8),%r14
0x7ffff7de76a8  <+0x0088>        0f 1f 84 00 00 00 00 00  nopl   0x0(%rax,%rax,1)
        72 [1]	in dl-init.c
0x7ffff7de76b0  <+0x0090>        4c 89 ea                 mov    %r13,%rdx
0x7ffff7de76b3  <+0x0093>        4c 89 e6                 mov    %r12,%rsi
0x7ffff7de76b6  <+0x0096>        89 ef                    mov    %ebp,%edi
0x7ffff7de76b8  <+0x0098>        ff 13                    callq  *(%rbx)
0x7ffff7de76ba  <+0x009a>        48 83 c3 08              add    $0x8,%rbx
        71 [2]	in dl-init.c
0x7ffff7de76be  <+0x009e>        4c 39 f3                 cmp    %r14,%rbx
0x7ffff7de76c1  <+0x00a1>        75 ed                    jne    0x7ffff7de76b0 <call_init+144>
        74 [1]	in dl-init.c
0x7ffff7de76c3  <+0x00a3>        5b                       pop    %rbx
0x7ffff7de76c4  <+0x00a4>        5d                       pop    %rbp
0x7ffff7de76c5  <+0x00a5>        41 5c                    pop    %r12
0x7ffff7de76c7  <+0x00a7>        41 5d                    pop    %r13
0x7ffff7de76c9  <+0x00a9>        41 5e                    pop    %r14
0x7ffff7de76cb  <+0x00ab>        c3                       retq
0x7ffff7de76cc  <+0x00ac>        0f 1f 40 00              nopl   0x0(%rax)
        45 [1]	in dl-init.c
// ...
```

It seems the segfault occurs due to boost (tested with boost 1.59).

I did some tests to narrow down the cause, and got following observations:
* The segfault disappears if all the OMPL related code is removed so that OMPL is not linked.
* The segfault occurs when `GeometricStateSpace : ::ompl::base::StateSpace` code is linked, and the segfault is persistent even though I remove all the code inside of the constructor of `::ompl::base::StateSpace`.

@mkoval @psigen Could you shed some light on this?

jslee02, 2018-04-06:
@dqyi11 The solution works for me, thanks! Also, I found this issue doesn't occur with Clang.

Let me post about this to the upstream hoping `BOOST_MATH_NO_LONG_DOUBLE_MATH_FUNCTIONS` is added to OMPL if there is no potential problem with it.

For the time being, let's disable GCC + OMPL (>= 1.2.0) in AIKIDO.

jslee02, 2018-04-07:
Update: This problem doesn't occur with newer GCC (at least 7.2.0). Presumably, this is a bug in GCC.

egordon, 2018-10-08:
Update on my end:

All tests pass on Ubuntu Bionic defaults:
OS: Ubuntu 18.04
GCC: 7.3.0 (bionic default)
Boost: 1.65.1.0 (bionic default)
ROS version: Melodic
OMPL: 1.4.0 (ros-melodic-ompl bionic package)

Wondering if the cmake check can also factor in GCC / Boost version.

brianhou, 2019-09-02:
We've decided to jump from Indigo to Bionic for longer-term stability, which also conveniently makes this not an issue for us. See potential fixes in #473 if you need to use Kinetic.

closed 2019-09-02 by brianhou