personalrobotics/aikido #365: Segment faults on 32bit machine due to Alignment issues from Eigen

dqyi11, 2018-04-05:
A few unit tests failed on 32bit machine due to segment faults. 
>The following tests FAILED:
>         14 - test_FrameDifferentiable (OTHER_FAULT)
>         21 - test_RnBoxConstraint (OTHER_FAULT)
>         22 - test_SE2BoxConstraint (OTHER_FAULT)
>         23 - test_RnConstantSampler (OTHER_FAULT)
>         25 - test_FiniteSampleable (OTHER_FAULT)
>         26 - test_CyclicSampleable (OTHER_FAULT)
>         33 - test_CartesianProductProjectable (OTHER_FAULT)
>         43 - test_BarrettHandKinematicSimulationPositionCommandExecutor (OTHER_FAULT)
>         52 - test_ParabolicTimer (OTHER_FAULT)
>         64 - test_VectorFieldPlanner (OTHER_FAULT)
>Errors while running CTest

They are caused by alignment issues from `Eigen`.
https://eigen.tuxfamily.org/dox/group__DenseMatrixManipulation__Alignement.html

dqyi11, 2018-04-20:
Resolved in PR #368

brianhou, 2018-04-21:
If I remember correctly, there are still alignment issues but @dqyi11 believes that they're in DART rather than AIKIDO at this point. If that's the case, maybe we should try to create a minimal test case that only uses DART to help with debugging.

I think we also concluded that since it's only (apparently) a problem on 32-bit machines, we shouldn't invest too much time into it. But now that @dqyi11 is really good at sniffing out those issues, it might be worth it for the two of you to set up a short debugging session to see if you can fix it quickly.

jslee02, 2018-04-21:
@brianhou Thanks for the recalling the story. In that case, I would like to leave this issue open until we actually figure out whether if this is AIKIDO's issue or not.