personalrobotics/aikido #384: IK Sampling

aditya-vk, 2018-04-26:
Snap Planners can potentially give out very long paths if the IK sampled is very far in the joint space from the current configuration. The sampling strategy should account for the current configuration and sample IKs minimizing the joint movements.

@jslee02 This should be easy to do with `DART` right? We should tie the sampler to an objective function or is there a simpler solution?

brianhou, 2018-04-26:
Please make this issue title more descriptive. Ideally, we should be able to read the title and have some idea of what this issue is about.

aditya-vk, 2018-04-30:
The problem stems from how `planToTSR` currently is. The seed values for the IK solver need to be more intelligently picked and not arbitrarily. If arbitrary is fine, then it doesn't hurt to start with the current configuration as the initial guess.

aditya-vk, 2018-05-05:
Solution discussed offline to address this correctly:
1. Create a Sequential Sampleable: https://github.com/personalrobotics/aikido/pull/393
2. API to make IKSampleable to return a set of IK solutions
3. Rank the samples obtained.

planToTSR takes in a SequentialSampleable with [FiniteSampleable(initialConfiguration), Sampleable]

brianhou, 2018-05-07:
I don't think (2) requires an API change: we can just keep calling `IkSampleable::sample` when trying to plan to a TSR.

PrPy defined a few methods for ranking that we should implement. The one that we probably care about the most (at least for this current issue) is [`NominalConfiguration`](https://github.com/personalrobotics/prpy/blob/master/src/prpy/ik_ranking.py#L55).

I think we can also just allow `planToTSR` to take in the uniform `Sampleable` over the configuration space (what it's probably using now) and then augment that inside of `planToTSR` before we pass that to `IkSampleable` by creating the `SequentialSampleable` as you described.

aditya-vk, 2018-12-04:
This has been addressed by #393 and #423 .

closed 2018-12-04 by aditya-vk