personalrobotics/aikido #405: Style: Adding empty line between header blocks or not

jslee02, 2018-05-09:
https://github.com/personalrobotics/aikido/pull/314#discussion_r187201620

brianhou, 2018-05-09:
Our current version of clang-format (3.8, the latest version available on Trusty) doesn't allow us to automatically handle this. Upgrading to Xenial should allow us to use clang-format-4.0.

We can set `IncludeBlocks` to be [`IBS_Regroup`](https://clang.llvm.org/docs/ClangFormatStyleOptions.html) to merge multiple includes into one block, sort, and then separate according to category.

aditya-vk, 2018-05-10:
Related issue for style on headers. Full path needs to be specified. 
For instance, this needs correction: https://github.com/personalrobotics/aikido/blob/master/include/aikido/statespace/dart/detail/JointStateSpaceHelpers-impl.hpp#L31

brianhou, 2019-08-29:
@madan96 Since we're updating clang-format to 6.0, I think we should be able to use this new option to automatically format header blocks. Can you try this out? (And simultaneously try to update the configuration so that we have minimal changes otherwise.)

@aditya-vk I think specifying the full path is a separate issue that can't be automatically fixed by clang-format. If we really care, I think we'll need someone to manually go through them.