personalrobotics/aikido #447: Locking the skeleton while planning

aditya-vk, 2018-07-14:
Previously the skeleton was locked in `planTo...` methods before calling any of the planners to plan. 

I think it should now reside in `<Planner><Problem> -> plan()` methods. @brianhou, can you confirm?

brianhou, 2018-07-15:
Hmm, I could be wrong but I was under the impression that @sniyaz tried to preserve the locking in the lowest level `planTo` functions (e.g. https://github.com/personalrobotics/aikido/blob/master/src/planner/vectorfield/VectorFieldPlanner.cpp#L151).

Regardless, I think this sounds reasonable. We might also want to consider adding an argument `bool lockSkeleton` to `Planner::plan` that controls whether the planner needs to lock the skeleton. This would be useful for wrapper/delegate planners: if the wrapper has already acquired the skeleton's lock in order to do something (e.g. sample configurations from a TSR), then the delegate doesn't need to lock the skeleton.

brianhou, 2019-02-02:
We should use DART's `MetaSkeleton::getLockableReference()` method rather than locking `Skeleton`s (available since DART 6.5).

sniyaz, 2020-10-16:
I believe that https://github.com/personalrobotics/aikido/pull/588 handles and documents this, at least for DART planners (I'm not sure what to do with non-DART planners, considering that those don't even know they're operating on a skeleton).

@brianhou @aditya-vk Does this seem like enough to close this issue, or are there other things we want to look at?