personalrobotics/aikido #449: Introducing setter methods in problem classes

aditya-vk, 2018-07-18:
Since the states are now ScopedStates as against raw pointers previously [see #442], if I am not wrong we would need setters. When the problem is instantiated it is not necessary that the goal is already set. Since the class creates its own unique pointer to hold the goal state, we would need a setter to access this location once the goal is determined [example: when planning to TSR].

@jslee02, @brianhou can you confirm this requirement or if I am missing something?

brianhou, 2018-07-18:
As @jslee02 said, I think we should only construct a problem once all the parts of the problem are known. If we're solving a different problem, we should construct an entirely new `Problem`. (In particular, I'd rather be strict about what kinds of problems can be created and solved. It doesn't make sense to provide a null goal.)

We only added `aikido/robot/util.cpp` to move that code out of `libherb`; the plan was to always use the `Planner` API once it was available. We should deprecate/delete that and migrate fully to the new API. [This](https://github.com/personalrobotics/aikido/blob/cfba2f6bd793190620e4058da7d4cc392dedefd8/src/planner/dart/ConfigurationToConfiguration_to_ConfigurationToTSR.cpp#L92) is how we currently implement the configuration sampling in the `Configuration->TSR` adapter.

To be clear, I'm not entirely opposed to setters for the start configuration, goal configuration/TSR/etc. I'm sure that constructing new problems is less efficient than modifying previous problems. However, I believe that including them could potentially open up a new variety of bugs for users (e.g. when we're parallelizing planning in MAGI).

I think keeping it immutable in this way makes it easier to reason about our code, so I'd rather continue without setters until we find a need for them. I'm open to revisiting this decision if we find that this approach doesn't actually work or is too inflexible/inefficient.

aditya-vk, 2018-07-19:
Sounds good. We should add a check for null start/goal states when the problem is created in that case to enforce 
> we should only construct a problem once all the parts of the problem are known.

Closing this for now.

closed 2018-07-19 by aditya-vk