personalrobotics/aikido #455: Mismatch in logmap representation of SE(2) states

aditya-vk, 2018-08-10:
AIKIDO assumes [\theta, x, y] while DART assumes [x, y, theta]. Everytime the the skeleton is set to a position the coordinates are set incorrectly.

One clean solution would be to have the state representation to be equivalent to that of DART, OMPL: 
[x, y, theta].

Is there any particular reason we chose this representation?

jslee02, 2018-08-10:
The general convention in DART is that angular terms come first followed by linear term like `FreeJoint`. `PlanarJoint` (SE(2) joint) complied the rule when it's created, and then for some reason ["I" changed the order](https://github.com/dartsim/dart/commit/9b596e296a401fa347cfc037085409a4d61a1c1e#diff-440599fc9e290b829676f5e30c7a436b), which I don't remember the reason. :disappointed: 

I personally prefer angular term first, but this is just a matter of convention. So I'm open to follow DART's convention or OMPL's as long as we're consistent.

aditya-vk, 2018-08-10:
We do not have to follow OMPL's convention since we never use their states anyway. It'd be better to stick to DART's convention IMO. I personally prefer angle at the end haha! Can we change it in DART then?

PR #458 (2018-10-23): Follow DART SE(2) convention
Resolves #455.

***

**Before creating a pull request**

- [x] Document new methods and classes
- [x] Format code with `make format`

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`
- [x] Add unit test(s) for this change
closed 2018-10-23 by brianhou