personalrobotics/aikido #555: Formatting checks disabled (possible erroneous?)

aditya-vk, 2019-10-02:
Two issues:

1. Format checks are currently disabled. I think this was intentional during the shift to bionic and clang-format-6.0. We need to bring it back.

I am not sure if the following issue is local to me (@madan96 can you please verify)
2. On running the formatting command, #521 seems to still fail. For instance `src/rviz/InteractiveMarkerViewer.cpp` Line 206 is never corrected on running clang-format. I have clang-format-6.0 on bionic.

madan96, 2019-10-02:
Yeah, I just checked and there seem to be some src files in `rviz`, `perception` and `control` subdirectories that still require formatting changes. Not sure how they got left out, there is no PR after #549 that was merged into master after this so I think that's where the problem is.

madan96, 2019-10-02:
@brianhou Should we run [check-format](https://github.com/personalrobotics/aikido/blob/master/.ci/script_catkin.sh#L17) for all builds or at least `BIONIC_CATKIN_FULL_RELEASE`?

brianhou, 2019-10-03:
We only need to check the formatting on one build.