personalrobotics/aikido #558: Expose custom velocityLimits in ConcreteRobot retimers

egordon, 2019-11-14:
Currently, all retiming functions just pull the velocity limits directly from the MetaSkeleton. To get around this, ADA currently sets the MetaSkeleton velocity limits manually, calls the retimer, and then resets them to the previous values.

Instead, this should be exposed as an optional argument (default is to use the MetaSkeleton), as there are times where we may want the robot to move more slowly for a specific trajectory.

See comments on `libada` PR40 for more details (https://github.com/personalrobotics/libada/pull/40).

brianhou, 2019-11-14:
I wonder if this makes sense to add as another possible field in the parameter struct for post-processors, which @sniyaz started working on in https://github.com/personalrobotics/libherb/pull/114. (Those struct definitions should eventually move to aikido.)

sniyaz, 2020-08-07:
Closed now that https://github.com/personalrobotics/aikido/pull/579 is merged.

closed 2020-08-07 by sniyaz