personalrobotics/aikido #563: ConcreteManipulator should be inheriting from ConcreteRobot?

gilwoolee, 2020-02-25:
Currently, 

- `Manipulator` inherits `Robot`
- `ConcreteRobot` inherits `Robot`
- `ConcreteManipulator` inherits `Manipulator`

the only difference between `Manipulator` and `Robot` is that `Manipulator` has `getHand` class. 
I think above inheritance makes sense conceptually, but code-wise, there's a lot of code duplicates between `ConcreteManipulator` and `ConcreteRobot`, esp. regarding TSR related methods.

I think we need to add `ConcreteRobot` to `ConcreteManipulator`'s parent class list, i.e. 
- `ConcreteManipulator` inherits `Manipulator` and `ConcretRobot`

gilwoolee, 2020-02-25:
@brianhou @aditya-vk @sniyaz @egordon  Any thoughts?

egordon, 2022-12-09:
Resolved by the Robot API overhaul (#593 )

closed 2022-12-09 by egordon